arduino-libraries/Arduino_NineAxesMotion #1: Sensor update time is slow

ajebulon, 2018-04-11:
I need to get all sensor values (Gyro, Accelerometer, Magnetometer, Euler, and Quaternion) at the highest update rate possible (100Hz). But, everytime I called the updateGyro(), updateAccel() and so on, it takes more than 10ms, thus I couldn't reach the 100 Hz, especially because I need to send those data through Serial port.

I update all sensor values inside the loop() in the Arduino sketch, and I simply print all the values to Serial comm using Serial.print(). Is there any good way to do this and to get 100 Hz rate?

trullock, 2019-11-18:
Sending the values over the Serial will take some time, so this will be slowing you down. You may get the speed you need without these (presumably) debug comms.

ajebulon, 2019-12-16:
In that case, how is the proper way to log the sensor data to my PC?

trullock, 2019-12-16:
Ah I didnt realise that was your requirement.
Having advanced further through this library myself since the original post, ive also found the refresh rate to be slower than I'd like.

The BNO055 says you can sample at 100Hz using the internal oscillator for the IMU, but only 20Hz for the Magnetometer using 9DOF mode: https://cdn-shop.adafruit.com/datasheets/BST_BNO055_DS000_12.pdf see 3.6.3

I'll get back to looking at this soon