arduino-libraries/Arduino_NineAxesMotion #27: Serial monitor is getting static accelerometer data, expected output values to change when moved

Moses0524, 2023-10-07:
Serial monitor is getting static accelerometer data, expected output values to change when moved

More info:
Installed Arduino_NineAxesMotion 1.1.2 on Arduino IDE
Connected Arduino 9 Axis Motion Shield onto an Arduino UNO R4 WiFi 1.0.4
Uploaded Accelerometer.ino example file with no compile errors

Not sure what I'm doing wrong.  Read about reported trullock error in 2019, I'm assuming this problem has been fixed in the current version.

Thank you in advance for the help.

per1234, 2023-10-08:
Hi @Moses0524. Thanks for your report. I am also able to reproduce the fault you describe using my "V9.0" [**9 Axis Motion Shield**](https://docs.arduino.cc/hardware/9-axis-motion-shield) on an [**UNO R4 WiFi**](https://docs.arduino.cc/hardware/uno-r4-wifi) board.

Please note that this library is no longer maintained, as documented in the readme:

https://github.com/arduino-libraries/Arduino_NineAxesMotion#readme

> Note: this library is deprecated and no longer maintained.

For this reason, there is a low chance that anyone from Arduino will investigate the problem or produce a fix.

My recommendation is for you to post on Arduino Forum:

https://forum.arduino.cc/c/hardware/uno-r4/uno-r4-wifi/189

I'm sure we'll be able to help you find a way to get the shield working with the UNO R4 WiFi over there.

Jorgetrujilloroman, 2024-11-27:
Hello, I found on the Forum that this line solves the issue:

[Forum Solution](https://forum.arduino.cc/t/arduino-uno-r4-mit-9-axis-motion-shield/1202556/9)